MicrosoftDocs/windows-driver-docs #2461: Broken links in WDK 8.1 secion

fenchelfen, 2020-08-27:
In "Step 2: Install the WDK" links to WDK 8.1 are dead

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3f5d6d55-cb35-6b9e-f8ca-e75f70826dc3
* Version Independent ID: 1d913516-b1e0-e872-c337-e30ed5dffce8
* Content: [Previous WDK versions and other downloads - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads#feedback)
* Content Source: [windows-driver-docs-pr/other-wdk-downloads.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/other-wdk-downloads.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

DOMARS, 2020-08-27:
@fenchelfen - Thanks for reporting this broken link, we are investigating.

RAJU2529, 2020-08-28:
@DOMARS , I checked now under 8.1 update, two links are broken and i could not able to find an alternate link

monrapps, 2020-09-03:
I have downloaded the wdksetup.exe from this link few days ago (that now is broken). but even the installer can't download the required files for the installation. Error message: "Setup cannot locate Windows Debugging VS Integration Dev12-x86_en-us.msi"

EliotSeattle, 2020-10-22:
**Note**   We removed the Windows 8.1 WDK downloads, because they do not meet our current requirements for downloadable files. They were signed by SHA-1 certificates, and we require SHA-2 certificates. As soon as we can rebuild and test these versions of the WDK, we will replace them.
Sorry for any inconvenience.
Eliot

closed 2020-10-22 by EliotSeattle

jafeucht, 2021-09-10:
So it's been almost a year since the Windows 8.1 WDK has been removed. Any plans to have it put back up with the correct hash, any time soon?

JustinBrow, 2022-07-12:
Hello, it's been almost 2 years since the Windows 8.1 WDK has been removed. Any plans to have it put back up with the correct hash, any time soon?

mirh, 2023-09-24:
RIP fd1086ba18d6ec4e80f3809317b7f835a73ece69