MicrosoftDocs/windows-driver-docs #2462: SD Bus Sample

DArk0711, 2020-08-27:
SD Bus Sample not found
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4639a65e-5ec8-2935-37fd-7e8e636f8546
* Version Independent ID: 24511b4a-afe6-ca90-dfda-855203d60af5
* Content: [SD Bus Driver Design Guide - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/sd/)
* Content Source: [windows-driver-docs-pr/sd/index.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/sd/index.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-08-27:
Thanks for bringing this to my attention, @DArk0711. It's likely that this sample was deprecated and the link should have been removed, but I'll look into it.

EliotSeattle, 2020-08-27:
As I suspected, the sample hasn't been updated for Windows 10. It's still available, though, and I've updated the text in the topic and the link:

**SD Bus Sample (Windows 8.1)**
This sample (no longer supported) is a functional Secure Digital (SD) IO driver. The driver is written using the Kernel Mode Driver Framework. It is a driver for a generic mars development board that implements the SDIO protocol without additional functionality.

Download the [Storage SDIO driver sample](https://github.com/microsoftarchive/msdn-code-gallery-microsoft/tree/master/Official%20Windows%20Driver%20Kit%20Sample/Windows%20Driver%20Kit%20(WDK)%208.1%20Samples/%5BC%2B%2B%5D-windows-driver-kit-81-cpp/WDK%208.1%20C%2B%2B%20Samples/Storage%20SDIO%20Driver/C%2B%2B) from GitHub.

The change should be live shortly.

Hope this helps.
Eliot

closed 2020-08-27 by EliotSeattle