MicrosoftDocs/windows-driver-docs #2464: Broken link in the sentence: "User-mode display drivers can also support the Direct3D version 10 DDI."

shaboinkin, 2020-08-27:
In this article, it mentions that the user mode driver needs to support DirectX 9 DDIs and provides a link to the page listing the interfaces.
The link for "Direct3D version 10 DDI" takes you here: https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/
I would have expected it to take me to a similar page as the DirectX 9 DDIs.

Could the link be fixed? 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b8543272-7482-e484-3294-292cbf6aab90
* Version Independent ID: 43074bab-879d-218b-7e8d-991dc6905f87
* Content: [User-Mode Display Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/user-mode-display-drivers)
* Content Source: [windows-driver-docs-pr/display/user-mode-display-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/user-mode-display-drivers.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

RAJU2529, 2020-08-28:
@lorihollasch . I am creating PR now.

RAJU2529, 2020-08-28:
@wdkbot  @lorihollasch , verify this PR #2466 and merge if it correct.

lorihollasch, 2020-09-25:
@shaboinkin , thanks for reporting this. And thanks to @RAJU2529 for the PR to fix it!

closed 2020-09-25 by lorihollasch