MicrosoftDocs/windows-driver-docs #2468: What's version 1912?

TomoharuMisawa, 2020-08-30:
it's Windows 10 version ?
i don't know version.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6ecebaac-b57c-75a0-88df-648b60d8f1cf
* Version Independent ID: 649d603f-72b9-fe61-3c20-89e3d44de4dd
* Content: [UDP Segmentation Offload (USO) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/udp-segmentation-offload-uso-)
* Content Source: [windows-driver-docs-pr/network/udp-segmentation-offload-uso-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/udp-segmentation-offload-uso-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

RAJU2529, 2020-08-31:
@TomoharuMisawa .It might be  1903 or 1909, but not  1912

aviviano, 2020-09-02:
Thanks @TomoharuMisawa for letting us know about this issue. I'm waiting for confirmation on the version (1903 or 1909), and will let you know as soon as I find out.

aviviano, 2020-09-16:
@TomoharuMisawa The answer is Windows 10 version 2004. The page is now updated. Thanks again.

closed 2020-09-16 by aviviano