MicrosoftDocs/windows-driver-docs #2469: broken links

16-Bit-Dog, 2020-08-30:
[Enter feedback here]
links seem broken on: Docs  /Windows  /Windows Drivers  /Driver Technologies  /HID / Interpreting HID Reports

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 55ff87a0-bf30-9d72-274f-e49378e35939
* Version Independent ID: 16b2af90-22ac-3f08-8ef7-2538de9860a9
* Content: [Interpreting HID Reports - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/interpreting-hid-reports)
* Content Source: [windows-driver-docs-pr/hid/interpreting-hid-reports.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/hid/interpreting-hid-reports.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

RAJU2529, 2020-08-31:
all most all the hyperlinks embedded in this article, redirects into same article

16-Bit-Dog, 2020-08-31:
is the redirection into the same article the intention of the hyperlink?

RAJU2529, 2020-08-31:
@16-Bit-Dog , Not that intension, authors and contributors will fix these issues very soon. Thanks for reporting

16-Bit-Dog, 2020-08-31:
thank you for the answer

RAJU2529, 2020-08-31:
@billlattimer . embedded hyperlinks are redirected to the same article , so please look into this issue.
Moreover, i could not able to find an alternate links.

EliotSeattle, 2020-08-31:
@billlattimer, source is available in .chm again. For this one, might make sense to add the topics here, each as an h2. They are all quite short. If I had time, I'd volunteer, but the links transformation this week has me jammed up.

billlattimer, 2020-09-18:
@16-Bit-Dog , @RAJU2529 thank you for calling this out. This content is being published today.

closed 2020-09-18 by billlattimer