MicrosoftDocs/windows-driver-docs #2470: Wrong UEFI specification section.

steven-bellock, 2020-08-31:
UpdateCapsule and QueryCapsuleCapabilities are in section 8.5.3 of the latest (2.8) specification, not 7.5.3.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1b4320a0-085e-4ca1-e83a-1b030b480100
* Version Independent ID: 5e474f7e-998e-1d76-49e7-fbc67ca1071d
* Content: [Windows UEFI firmware update platform - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/windows-uefi-firmware-update-platform)
* Content Source: [windows-driver-docs-pr/bringup/windows-uefi-firmware-update-platform.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/windows-uefi-firmware-update-platform.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

RAJU2529, 2020-09-03:
@steven-bellock . Thanks for pointing out . I  downloaded  uefi specification 2.8 and i  verified  UpdateCapsule and QueryCapsuleCapabilities cotent in ebook , and i found out , what you said is correct.

barrygolden, 2020-09-10:
Thanks @steven-bellock for filing this issue!

Thanks to @RAJU2529 for doing the research and submitting PR #2478!

closed 2020-09-10 by barrygolden