MicrosoftDocs/windows-driver-docs #2479: SDEL Tokens link does not go anywhere

rahul-ramadas, 2020-09-03:
The link to "SDEL Tokens" doesn't go anywhere and I can't find an appropriate page for it.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d87975db-4d2d-8ec4-8d18-fd6751c1b4b8
* Version Independent ID: 56d486a3-369a-db10-5646-dccfad6f3aa6
* Content: [Simple Data Evaluation Language Overview - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdtf/simple-data-evaluation-language-overview)
* Content Source: [windows-driver-docs-pr/wdtf/simple-data-evaluation-language-overview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdtf/simple-data-evaluation-language-overview.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

RAJU2529, 2020-09-03:
@rahul-ramadas , i also checked, SDEL tokens link does not exist

EliotSeattle, 2020-09-03:
Thanks for alerting us to this, @rahul-ramadas. Looks like we lost some pages in a content migration a few years ago. I am reconstructing the pages and hope to have them published later today.
Eliot

EliotSeattle, 2020-09-09:
Verified that these are live.
Thanks!!!

closed 2020-09-09 by EliotSeattle