MicrosoftDocs/windows-driver-docs #2480: Netmon Tool Download broken and not accesible

somajagadeesh, 2020-09-03:
[Network Monitor Tool download link is broken on the URL https://docs.microsoft.com/en-us/windows-hardware/drivers/portable/using-the-netmon-tool which is being shared to customers and CSS]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0c542807-0787-c386-43d7-79f230e23bcd
* Version Independent ID: beec032f-ad86-4615-3702-c243454f3eb8
* Content: [Using the Network Monitor Tool - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/portable/using-the-netmon-tool)
* Content Source: [windows-driver-docs-pr/portable/using-the-netmon-tool.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/portable/using-the-netmon-tool.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-09-03:
Thanks for bringing this to my attention, @somajagadeesh 
I believe that NetMon is deprecated, replaced by Message Analyzer. But I will verify one way or the other and get back to you.
Eliot

somajagadeesh, 2020-09-04:
Below link says Message Analyzer is retired or deprecated 
https://docs.microsoft.com/en-us/message-analyzer/installing-and-upgrading-message-analyzer

EliotSeattle, 2020-09-22:
The tool has been reinstated and is available for download once again.
Thanks so much, @somajagadeesh!!!
Eliot

closed 2020-09-22 by EliotSeattle