MicrosoftDocs/windows-driver-docs #2481: Hyperlink issue

dcRUSTy, 2020-09-04:
[Enter feedback here]

Line Number 13 in .md file leads to hyperlink as https://windows-hardware/drivers/debugger/
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 142adea7-b7b8-4a5a-5e19-83a202ab04b2
* Version Independent ID: 489c0c22-908c-aa93-520c-836c941b5b4e
* Content: [Getting Started with WinDbg (User-Mode) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/getting-started-with-windbg)
* Content Source: [windows-driver-docs-pr/debugger/getting-started-with-windbg.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/getting-started-with-windbg.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2020-09-04:
@dcRUSTy , I am fixing the hyperlink now.

dcRUSTy, 2020-09-04:
@RAJU2529 not very sure hardcoding the domain name was required or not.... removing one `/` from `//windows-hardware/drivers/debugger/` is also one of the solution

RAJU2529, 2020-09-04:
@dcRUSTy , also share the screenshot, location of **Line Number 13 in .md file leads**

EliotSeattle, 2020-09-04:
Corrected in source and merged to main branch. Should be live momentarily.

Thanks for bringing this to our attention, @dcRUSTy 
Eliot