MicrosoftDocs/windows-driver-docs #2486: All 3 links on page are broken and produce 404 not found

exedor, 2020-09-07:
All the links on the page, specifically the following:
"OEMForceFeedback Registry Settings"
"Driver Interface"
"User-Mode Functions"

produce a 404 Not Found page when followed/clicked.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7231b7cd-29bb-6501-c0ca-acdd5c9d4a70
* Version Independent ID: fdbbf990-6adc-9b66-3714-c2173f8b13f9
* Content: [Force Feedback Device Driver Interface - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/force-feedback-device-driver-interface)
* Content Source: [windows-driver-docs-pr/hid/force-feedback-device-driver-interface.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/hid/force-feedback-device-driver-interface.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

EliotSeattle, 2020-09-09:
Dropped in migration.

exedor, 2020-09-13:
Not sure how to interpret that. It contains extremely little info. Was that intentional? If so, you should probably remove the links. Was it not intentional, i.e. accidentally dropped in migration? Will they be restored and corrected?

billlattimer, 2020-09-13:
@exedor that was a comment to me as the topic owner regarding the source of the missing docs; I'll be making this a priority over the next few days to restore.

exedor, 2020-09-13:
Ahh, much appreciated. Thank you!

billlattimer, 2020-11-06:
@exedor thanks again for calling this out - the content that was in the linked articles was moved into the main article and is now available. 

https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/force-feedback-device-driver-interface

closed 2020-11-06 by billlattimer

cyberluke, 2022-08-21:
Lol guys half of internet reference that URL. At least learn to code website and provide HTTP 302 redirect, damn!