MicrosoftDocs/windows-driver-docs #2487: ProjectUpgradeTool.exe still referenced in doc tree despite not shipping with WDK

duvallj, 2020-09-07:
I was attempting to update a driver that used the old `build.exe` system to be compatible with the newer `MSBuild` process when I found this doc page. <https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/index-of-windows-driver-kit-tools> does not list ProjectUpgradeTool.exe as a WDK tool (or even at all), nor I could also not find the tool when searching my local installation of the WDK, so I am assuming the documentation on this page is out-of-date.

In my mind, there are two ways to resolve this issue: either by including ProjectUpgradeTool.exe in the WDK again and updating the index page (which I would very much like, but is off-topic here), or by adding a warning on this page that the tool is no longer included with the latest version of the WDK.

Thank you,

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 880e65ab-4a83-5c10-cb07-dd1313ec9094
* Version Independent ID: cf9fd5ed-ab04-205a-8b9d-5244463878ac
* Content: [ProjectUpgradeTool - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/projectupgradetool)
* Content Source: [windows-driver-docs-pr/devtest/projectupgradetool.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/projectupgradetool.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

DOMARS, 2020-09-10:
@duvallj - Thanks for pointing out what you are seeing. 

In case it is helpful, you could consider downloading a previous version on the WDK and see if the tool is included in an older download. They older WDK downloads should be here: https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads

duvallj, 2020-09-19:
@DOMARS thank you for responding! I apologize for getting back to you so late (school has been crazy)

The version of the WDK most likely to contain ProjectUpgradeTool.exe, that is WDK 8.1 Update, is a broken link on that page (<https://go.microsoft.com/fwlink/p/?LinkId=393659> goes nowhere), though that is a separate issue.

As luck would have it, though, WDK for Windows 10, version 1607 still contains the tool! If I look through the rest of the kits some more, would it be appropriate for me to introduce a PR detailing when exactly this tool seemed to be removed?

DOMARS, 2020-09-21:
@duvallj - I'm glad you were able to locate a copy of the tool. If you are able to see the last kit that the ProjectUpgradeTool.exe was included, that would be a useful data point. We would work to confirm that before adding wording indicating that it is no longer available after that version and acting on a PR. In general I would use a version of the SDK/WDK to match the age of the older VS project. And of course work to bring it to current day versions for the best compatibility, security and reliability.

tedhudek, 2020-10-22:
Thanks for bringing this to our attention @duvallj.  You're right, the tool no longer ships with the kit and is unsupported. I've added a note to the top of the doc page - should be up on the live site in a few minutes

closed 2020-10-22 by tedhudek