MicrosoftDocs/windows-driver-docs #2490: Missing parameter 1 (0x00000000000000f0) in Bug Check 0xA0: INTERNAL_POWER_ERROR

miikaah, 2020-09-08:
>The computer has rebooted from a bugcheck.  The bugcheck was: 0x000000a0 (0x00000000000000f0, 0x0000000000000004, 0x000000000000000e, 0xffffb38360639040). A dump was saved in: C:\WINDOWS\MEMORY.DMP. Report Id: 6e18e100-febe-482b-b766-8bcffdd6164b.

There is no section for 0x00000000000000f0. The parameters are always the same except for the last one which I assume is some sort of a checksum or size of _something_.

The computer crashes while in sleep mode or going to sleep mode or waking up from sleep mode. Sometimes going to sleep the OS shuts off peripherals but the computer keeps running. It is random but happens often enough to question why even use sleep mode or Windows.

There's about a million things to debug without more specific documentation including (but no limited to):
- Power supply
- CPU max heat threshold (nope)
- RAM
- Outdated drivers (nope, fresh reinstall of Windows 10 too)
- Turning off Fast Startup
- Various power options that might disallow the computer going to sleep like media streaming (nope)
- Turning on Hibernation via command line (let's hope this one is the winner!)
- Peripherals
- etc

I had this same problem a year a go and since the computer was about 9 years old I thought it must be corrupted memory or something, which made me buy new components. So imagine my delighted surprise that the same issue has started happening again making me question why even live. What's the point. I don't want to be tech support to myself for the rest of my life. I don't want to be writing issues to somebody that nobody will probably read nor care to fix. 

Could you please for the love of Ukko improve this part of your documentation to point me in the right direction. I mean a few weeks a go I reconnected an old LaCie USB external drive (doesn't even have power) is that the reason? Is Mars in retrograde? How was this computer working for a year without incident yet now suffers from the exact issue I had a year a go with a mostly different computer. It's just wild.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: eb666216-10cc-bf5c-5856-412c08d035ff
* Version Independent ID: 1e42770f-2b01-bf39-e026-5ac39dfd3408
* Content: [Bug Check 0xA0 INTERNAL_POWER_ERROR - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xa0--internal-power-error)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xa0--internal-power-error.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xa0--internal-power-error.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

miikaah, 2020-09-22:
Cause was most likely an outdated Startup program that started failing after a Windows Update, VirtualCloneDrive I'm looking at you.

miikaah, 2020-10-22:
Nope still happens. Current uptime is around 9 days and then... BOOM.

DOMARS, 2020-12-09:
@miikaah - Thanks for the input, I updated the docs to include the 0XF0 stop code. In general the OEM that selected the hardware and drivers would be the one to use a kernel debugger to look in the faulting source code to determine why the code is failing. Without access to the source code, you can look in the event log to see if a specific driver, or software program logged information before the blue screen occurred. If a specific program or hardware device seems to be indicated, you and attempt to remove or update the faulting element. Resetting the OS is another way to attempt to remove drivers or software that may be causing this issue. For general guidance, this topic may be of some help: https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/blue-screen-data Good luck with determining a solution to you re-occurring bug check.

closed 2020-12-09 by DOMARS

davidhcefx, 2022-04-11:
Thanks for the updated documentation!
Just wanted to add an additional info that after encountering `bugcheck 0x000000a0 (0x00000000000000f0, ...)`, I failed to find the file `C:\WINDOWS\MEMORY.DMP` as mentioned in the event logger. After googling around, I realized that they are located under `C:\Windows\Minidump` instead. Using [Nirsoft's BlueScreenView](https://www.howtogeek.com/196672/windows-memory-dumps-what-exactly-are-they-for/) can find out the causing driver.