MicrosoftDocs/windows-driver-docs #2500: broken link to message analyzer download page

aviwert, 2020-09-13:
I want to practice the message analyzer as described in the page but the link to the download leads to "we are sorry..." page
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b034fd72-470b-76e3-e080-46cb688242db
* Version Independent ID: 3fde5448-65db-8332-2194-c3b301d0d4d5
* Content: [Capture and view USB traces with Microsoft Message Analyzer - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-)
* Content Source: [windows-driver-docs-pr/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

RAJU2529, 2020-09-14:
@aviwert , message analyzer application is removed from the  Microsoft download server, because the application is no more produced.
But you can download from third party website

billlattimer, 2020-09-18:
@aviwert, MMA has been deprecated and this content was left online for reference only. It's now been archived.

closed 2020-09-18 by billlattimer