MicrosoftDocs/windows-driver-docs #2502: Minor typo on KSPROPERTY_JACK_SINK_INFO page

rhymu8354, 2020-09-14:
KSJACK_SINK_INFORMATION is misspelled as "KSJACK_SINK_INFORMAITON" ('I' and 'T' swapped in "information") in the Usage Summary Table section, right after the table.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b5e8652c-cce8-332b-ea54-907395e9a7d7
* Version Independent ID: 784d72a2-2a30-9d62-60ac-9fc5c31cab85
* Content: [KSPROPERTY\_JACK\_SINK\_INFO - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/ksproperty-jack-sink-info)
* Content Source: [windows-driver-docs-pr/audio/ksproperty-jack-sink-info.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/ksproperty-jack-sink-info.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2020-09-15:
@rhymu8354 . thanks for pointing out. Error will be fixed Within a week

DOMARS, 2020-09-15:
@rhymu8354 - We do like to have structure names properly spelled. So thanks for calling this out. And thanks to @RAJU2529 for creating a PR to fix it.

closed 2020-09-15 by DOMARS