MicrosoftDocs/windows-driver-docs #2508: No proper documentation for DPC watchdog?

DRSAgile, 2020-09-15:
Hi,

I am seeing bits of information about the watchdog (HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Watchdog\) in [some places](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/disabling-the-watchdog-timer-while-testing-display-drivers) but not as a complete guide.

For example, I can not find information on how to turn DPC watchdog off not just for display drivers as in the linked material but in general. Also, on how to change DPC time allotment (in ticks). (I have an old motherboard with the latest drivers but they are not going to be updated, of course, so my only solution to get rid of random 0x133 BSODs is to those two options).

But maybe I somehow failed to find documentation on this?

Thanks in advance.

EliotSeattle, 2020-09-16:
Thanks for reaching out @DRSAgile 
With certain rare exceptions, we do not advise adjusting or turning off watchdog timers. A watchdog is implemented to protect both the computer and the consumer.
Eliot

closed 2020-09-16 by EliotSeattle

DRSAgile, 2020-09-17:
I know it is not recommended, but it is necessary sometimes. 

More interestingly, does anyone knows at all where you can change the five hundred ticks threshold? I might not need to turn off the watchdog completely if I will be able to adjust the time allotment a bit.

Thanks in advance.

EliotSeattle, 2020-09-17:
Hi @DRSAgile.  
For safety reasons, we do not advice adjusting the watchdog timers. You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The communities there are quite responsive. 
[Microsoft Q&A](https://docs.microsoft.com/en-us/answers/products/) 
[OSR NTDEV forum](http://www.osronline.com/cf.cfm?PageURL=showlists.cfm?list=NTDEV) 
 
Again, thanks for reaching out. Best of luck with this issue.