MicrosoftDocs/windows-driver-docs #2510: WinUSB application template unusable

shrddr, 2020-09-16:
Windows 10.0.18362
VS 2019 16.7.3
WDK 10.0.19030.1000

https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/how-to-write-a-windows-desktop-app-that-communicates-with-a-usb-device
The procedure listed for _Single computer setup_ generates an untrusted .cat file, which makes installing the driver impossible. You can see the reason in `c:\Windows\INF\setupapi.dev.log` after a failed driver install attempt:

```
!    sig:                Verifying file against specific (valid) catalog failed.
!    sig:                Error 0x800b0109: A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.
```

Also if you live in GMT+X timezone and Build the project after midnight there's an issue with DriverVer which is solved by setting INF Package project settings > Inf2Cat > General > Use Local Time > Yes.

RAJU2529, 2020-09-18:
@shrddr , Which article are you referring?.
first add root certificate to the trusted publisher in certmgr

tedhudek, 2021-02-24:
Thanks @shrddr for the feedback and apologies for the delay. I'm not very familiar with .cat files - what was your workaround to get a trusted .cat file? Happy to add the caveat to the doc page.  Also you could feel free to spin up a PR, that might be easiest.

tedhudek, 2021-03-08:
Closing out due to lack of additional info. Would love to add this to the page, feel free to open a PR as needed

closed 2021-03-08 by tedhudek