MicrosoftDocs/windows-driver-docs #2516: Dead Links in Other WDK downloads

godfreychung, 2020-09-21:
https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads

These 2 links return error 404.

[WDK 8.1 Update](https://go.microsoft.com/fwlink/p/?LinkId=393659) (English only)
[WDK 8.1 Update Test Pack](https://go.microsoft.com/fwlink/p/?LinkID=393660) (English only)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3f5d6d55-cb35-6b9e-f8ca-e75f70826dc3
* Version Independent ID: 1d913516-b1e0-e872-c337-e30ed5dffce8
* Content: [Previous WDK versions and other downloads - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads)
* Content Source: [windows-driver-docs-pr/other-wdk-downloads.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/other-wdk-downloads.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @windows-driver-content
* Microsoft Alias: **windowsdriverdev**

EliotSeattle, 2020-09-25:
Thanks, @godfreychung 
I apologize for the missing downloads. We're aware of this and are actively working on a solution.
Eliot

EliotSeattle, 2020-10-22:
**Note**  We removed the Windows 8.1 WDK downloads, because they do not meet our current requirements for downloadable files. They were signed by SHA-1 certificates, and we require SHA-2 certificates. As soon as we can rebuild and test these versions of the WDK, we will replace them.
Sorry for any inconvenience.
Eliot

closed 2020-10-22 by EliotSeattle