MicrosoftDocs/windows-driver-docs #2518: Not clear on minimum requirements are for a Windows 10 system

aguptadel, 2020-09-22:
It’s not really clear what the minimum requirements are for a Windows 10 system. There are links in this section when mention stuff like WDDM 2.0 and Windows 10, or WDDM 1.2 for Windows 8. 

I guess a better question would be "what is the bare minimum that is needed to get a DirectX 9 GPU driver working in Windows 10?"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2e05c0fc-2bb3-964a-01e8-0e9b2f7557de
* Version Independent ID: 65380793-9c19-edbb-3879-f3dcbb8e15d4
* Content: [Migrating to the Windows Display Driver Model (WDDM) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/migrating-to-the-windows-vista-and-later-display-driver-model)
* Content Source: [windows-driver-docs-pr/display/migrating-to-the-windows-vista-and-later-display-driver-model.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/migrating-to-the-windows-vista-and-later-display-driver-model.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

RAJU2529, 2020-09-26:
@aguptadel . minimum requirement is WDDM 1.0 for windows 10 . but newer version is WDDM  2.8

lorihollasch, 2020-10-20:
@aguptadel , closing this per our email, thanks!

closed 2020-10-20 by lorihollasch