MicrosoftDocs/windows-driver-docs #2521: Definition for MAT/MADT?

fakecv, 2020-09-25:
There is a line start with MAT/MADT in "Firmware WEG terminology":
==========
MAT/MADT – Memory A
==========

I think something missed.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c409363e-8fcb-9c0b-07f8-72ff1620385c
* Version Independent ID: 58241802-551c-edb0-f562-6aa2a7f34507
* Content: [Firmware Windows Engineering Guide (WEG) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/firmware-weg)
* Content Source: [windows-driver-docs-pr/bringup/firmware-weg.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/firmware-weg.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

RAJU2529, 2020-09-25:
@fakecv . I am fixing now.

fakecv, 2020-09-25:
@RAJU2529  Thank you!

closed 2020-09-25 by fakecv

RAJU2529, 2020-09-26:
@nam31.  @e0i,  loot in to this PR #2523

RAJU2529, 2020-09-29:
@barrygolden . I  want to add more abbreviations,I collected sources from the intel website and from Wikipedia website.