MicrosoftDocs/windows-driver-docs #2522: There is a typo in the Receiving Network Data Topic of Introduction to Network Drivers document.

vanilet, 2020-09-25:
The first line of text right below the figure is "Miniport drivers call the NdisMIndicateReceiveNetBufferLists function to indicated NET_BUFFER structures to higher level drivers.".

It seems that there is a typo or missing verbs in the text near "... NdisMIndicateReceiveNetBufferLists function to indicated NET_BUFFER structures ..."

I think the verb 'to indicated' should be changed to 'to indicate'. As a result, the complete sentence will be "Miniport drivers call the NdisMIndicateReceiveNetBufferLists function to indicate NET_BUFFER structures to high level drivers".

Please review this sentence.

Thanks.

---
#### 문서 세부 정보

⚠ *이 섹션을 편집하지 마세요. docs.microsoft.com ➟ GitHub 문제 연결에 필요합니다.*

* ID: eeb5095d-52ac-428e-bc75-22592bf1ecd1
* Version Independent ID: ba0cc302-636f-c6bf-7571-3de82c8e46e5
* Content: [Receiving Network Data - Windows drivers](https://docs.microsoft.com/ko-kr/windows-hardware/drivers/network/receiving-network-data)
* Content Source: [windows-driver-docs-pr/network/receiving-network-data.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/receiving-network-data.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

RAJU2529, 2020-09-28:
@vanilet, now check the article again , typo error is  fixed already, If you confirm it , then c,lose this issue

vanilet, 2020-09-28:
> @vanilet, now check the article again , typo error is fixed already, If you confirm it , then c,lose this issue

I’ve checked the sentence. Thanks.

closed 2020-09-28 by vanilet