MicrosoftDocs/windows-driver-docs #2525: Broken WDK support for multiple Windows 10 versions in Visual Studio

wmjdgla, 2020-09-28:
I'm trying to develop and build a driver for multiple versions of Windows 10, from 1703 to 1903.

The "Previous WDK versions and other downloads" page states that a specific version of Visual Studio can only support specific versions of WDK. e.g. VS2017 supports WDK for Windows 10, 1709, 1803, 1809. This seems to imply that VS2017 can simultaneously support multiple versions of WDK.

So I installed VS2017, then installed WDK for Windows 10, 1709, and then 1803 to see how things would work out. After creating an empty WDM project and checking out its project properties, I found that there is no way to switch the WDK version used. The $(DDK_LIB_PATH) macro is set to use the latest installed WDK. I then installed VS2019 and then installed WDK for Windows 10, 1903. Returning to my VS2017 project, the $(DDK_LIB_PATH) has been updated to the path for 1903. 

So it looks like I have to use multiple VMs, each with its own VS and WDK to develop and build a driver for multiple versions of Windows 10. The "Other WDK downloads" mentioned nothing about this and the way things are now makes for awful development experience.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3f5d6d55-cb35-6b9e-f8ca-e75f70826dc3
* Version Independent ID: 1d913516-b1e0-e872-c337-e30ed5dffce8
* Content: [Previous WDK versions and other downloads - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/other-wdk-downloads)
* Content Source: [windows-driver-docs-pr/other-wdk-downloads.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/other-wdk-downloads.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-10-17:
Hi @wmjdgla.

Thanks so much for raising this. Current versions of the WDK overwrite the props, targets, and tool set, and these settings are left when the newer version is uninstalled. To rebuild the older WDK a new installation of that older WDK needs to happen. We are testing a solution to make sure it works in all edge cases, but for now here's what we recommend:

Leverage the EWDKs-as they are self contained-where you can. A bonus with using the EWDK is that you can launch Visual Studio from within the EWDK and that instance of VS will use the appropriate toolset from the EWDK it was launched from. 

One thing to note is that that the EWDK needs to match the VS major version installed. 

When we get things finalized, we'll update this topic with the official approach.

Hope this helps.
Eliot

closed 2020-10-17 by EliotSeattle

EliotSeattle, 2020-10-19:
The content is now on the download page, [https://docs.microsoft.com/windows-hardware/drivers/download-the-wdk](https://docs.microsoft.com/windows-hardware/drivers/download-the-wdk):

You can use the Visual Studio interface with the build tools provided in the EWDK.

1. Mount the EWDK ISO.
2. Run `LaunchBuildEnv.cmd`.
3. In the environment created in step 2, type **SetupVSEnv**, and then press **Enter**.
4. Launch devenv.exe from the same environment, using the full file path. Example: `C:\Program Files (x86)\Microsoft Visual Studio\2019\\%Community|Professionial|Enterprise%\Common7\IDE\devenv.exe`

Note that the Visual Studio major version should match with the version in the EWDK. For example, Visual Studio 2019 works with the EWDK that contain VS16.X build tools.

cyberluke, 2022-09-03:
Not good enough solution. What if you need to rebuild kernel driver for Windows 7.1, 8.1 and 10?