MicrosoftDocs/windows-driver-docs #2531: microsolf edge

stephane439, 2020-10-03:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cf05a7bc-42b9-118a-1d7a-40535d0c21ca
* Version Independent ID: 48713b46-3ce3-0d16-d9f2-22c04c6761c2
* Content: [Registering Miniport Driver Functions for WDM Lower Edge - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/registering-miniport-driver-functions-for-wdm-lower-edge)
* Content Source: [windows-driver-docs-pr/network/registering-miniport-driver-functions-for-wdm-lower-edge.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/registering-miniport-driver-functions-for-wdm-lower-edge.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

RAJU2529, 2020-10-05:
@stephane439 . Whats your query in this article?

EliotSeattle, 2020-10-05:
Thanks, @stephane439 for using the Issue reporting tool for the repo. I don't see any question about Microsoft Edge or any driver problems, so I am going to close this out. If you do have feedback or a question, please leave details in the Issue.

Thanks,
Eliot

closed 2020-10-05 by EliotSeattle