MicrosoftDocs/windows-driver-docs #2532: Device driver does not install on any devices, use primitive driver if this is intended

LusKrew, 2020-10-03:
Not sure what was the need for this change in visual studio, and i don't want to revert my version back..

how am i suppose to change the INF file to build a simple hello world ..?

```
;
; HelloWorld.inf
;

[Version]
Signature="$WINDOWS NT$"
Class=System
ClassGuid={4d36e97d-e325-11ce-bfc1-08002be10318}
Provider=%ManufacturerName%
DriverVer=
CatalogFile=HelloWorld.cat
PnpLockDown=1

[DestinationDirs]
DefaultDestDir = 12

[SourceDisksNames]
1 = %DiskName%,,,""

[SourceDisksFiles]

[Manufacturer]
%ManufacturerName%=Standard,NT$ARCH$

[Standard.NT$ARCH$]

[Strings]
ManufacturerName="<Your manufacturer name>" ;TODO: Replace with your manufacturer name
ClassName=""
DiskName="HelloWorld Source Disk"
```

RAJU2529, 2020-10-05:
@LusKrew ,  I would recommend you to post your question on  **https://developercommunity.visualstudio.com/spaces/8/index.html** 
so that experienced developers can answer your questions.

tedhudek, 2020-10-13:
Sorry, I don't know.

I'd recommend asking in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

Thanks for reaching out and best of luck with this issue.

closed 2020-10-13 by tedhudek