MicrosoftDocs/windows-driver-docs #2533: Spectre error even after i DISABLED spectre Mitigation in options.. : Spectre-mitigated libraries are required for this project. Install them..

LusKrew, 2020-10-03:
WHY am i getting Spectre mitigation and VS asking me to install them when i DISABLED the spectre mitigation in the options and just want to compile a SIMPLE HELLO WORLD? 

`Spectre-mitigated libraries are required for this project. Install them from the Visual Studio installer (Individual components tab) for any toolsets and architectures being used. Learn more: https://aka.ms/Ofhn4c`

RAJU2529, 2020-10-05:
@LusKrew , I would recommend you to post your question on https://developercommunity.visualstudio.com/spaces/8/index.html
so that experienced developers can answer your questions.

EliotSeattle, 2020-10-05:
Hi @LusKrew.

The [WDK download page](https://docs.microsoft.com/windows-hardware/drivers/download-the-wdk) has this info:

"WDK has Spectre mitigation enabled by default but requires spectre mitigated libraries to be installed with Visual Studio for each architecture you are developing for. Additionally, developing drivers for ARM/ARM64 require the build tools for these architectures to also be installed with Visual Studio. To locate these items you will need to know the latest version of MSVC installed on your system." 

Does that help?
Eliot

closed 2020-10-07 by EliotSeattle