MicrosoftDocs/windows-driver-docs #2552: "Windows Server 2008 or later"

GabeSJones, 2020-10-16:
I want to make sure that the information in the "How do production signing options differ by Windows Version?" is clear.  There was much confusion previously about whether WHQL would be required for Windows Server 2016 and 2019.  The final answer was that, no, Attested Signing was OK for these SKUs.  This table, however, says that for "Windows Server 2008 and later" the only option is "WHQL or drivers cross-signed before July 1, 2021."  I assume that the "and later" here really means "and 2012," and that Windows 10-based Server versions (2016, 2019, and future) are included in the "Windows 10" row instead.  I hate assuming, though, so could you make it explicit here that the newer versions of Windows Server are subject to the Windows 10 policy?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0e326d3e-833a-e312-d35d-11569690c7e6
* Version Independent ID: 7a9795cf-093e-a67f-8162-b832e1161ba8
* Content: [Deprecation of Software Publisher Certificates, Commercial Release Certificates, and Commercial Test Certificates - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates)
* Content Source: [windows-driver-docs-pr/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

nasantia, 2020-10-16:
Can you point me to the conflicting documentation of "The final answer was that, no, Attested Signing was OK for these SKUs."?
I can reach out to the author and clarify the confusion. My understanding was that attested signing was NOT OK for these SKUs

tedhudek, 2020-10-22:
@GabeSJones any comments on this one?  I'll close this out soon if there's no doc work to be done.

GabeSJones, 2020-10-22:
Sorry, I have intended to get back to this, but I haven't been able to dig
up the requested content yet.  I'll try to get back to you by tomorrow.

GabeSJones, 2020-10-23:
Attested signing has always worked for Windows Server 2016 (and 2019?  I haven't tested it as much, so I don't remember details there), but the documentation always stated that it would not work.  Apparently Peter @ OSR finally got a Microsoft PM to tell him that it will work. Thread here:  https://github.com/MicrosoftDocs/windows-driver-docs/issues/1068.  That page was updated late last year, though it still makes note of application only to "client" systems, and there are still plenty of pages that say it won't work at all.

There may be a difference here regarding "supported/not supported" vs "will/won't work."  (This page, for example, makes that distinction for Win7/8.1 vs Windows Server: https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/attestation-signing-a-kernel-driver-for-public-release)

tedhudek, 2020-10-28:
Thanks for the feedback on this one.  Closing it out since there is no doc change to be made

closed 2020-10-28 by tedhudek