MicrosoftDocs/windows-driver-docs #2555: Referenced audio driver sample SB16 doesn't seem to exist

mr-intj, 2020-10-21:
In the first paragraph...

> The SB16 sample audio driver in the Windows Driver Kit (WDK) shows how to use the PcRegisterPhysicalConnection routine to register the physical connections between subdevices that are contained in the same audio adapter.

I can't find any trace of the SB16 audio driver example in the current (or Windows 8.1 or 8.0) WDKs.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4fd8f772-e3ec-e2f0-1876-31129d8fae01
* Version Independent ID: dc17eab3-416a-cf02-0d87-74cb65b6df9b
* Content: [Driver Support for Dynamic Subdevices - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/driver-support-for-dynamic-subdevices)
* Content Source: [windows-driver-docs-pr/audio/driver-support-for-dynamic-subdevices.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/driver-support-for-dynamic-subdevices.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DJm00n, 2020-11-17:
Maybe  SB16 sample audio driver link could be changed to [SysVAD Virtual Audio Device Driver Sample](https://github.com/microsoft/Windows-driver-samples/blob/28f8fc7638b9149e05a7b15ec11cc08f9e6d7e86/audio/sysvad/common.cpp#L1980).

DOMARS, 2020-12-23:
@DJm00n - Thanks, yes this reference is now updated to Sysvad. I also worked to remove other references to SB16 in the audio docs, as most relevant current day techniques are now shown in Sysvad. Thanks @mr-intj for filing this git issue, and good luck with your development work.

closed 2020-12-23 by DOMARS