MicrosoftDocs/windows-driver-docs #2558: Windows 7/8: Usermode files signed with pagehashes [/codeIntegrity linked files]

pmisik, 2020-10-23:
Hi,

From my testing it seems expiration of cross-signed root certificates on now Windows 10 machines will affect user mode code that must be signed with page hashes (code that is linked with /codeIntegrity = PE files with set Check integrity flag in DLL characteristics). Like code interacting with Security center and other.

We do more than 4 releases a day with giant amount of dlls linked with /codeIntegrity flag. Rough underestimated calculation say we release more than 2GB of PE files/a day. Our biggest PE file has signature almost 2MB due to page hashes. On daily base we face known issue with “bad markers” related to MSRC case MS12-024 escalated through paid support with only solution to rebuild of dll pray new PE file do not hit this issue. The bigger PE file is bigger probability we hit this issue.

Unfortunately, significant amount of our customers still run Windows 7.
Microsoft Code Verification Root expires in ‎November ‎1, ‎2025. I understand it is SHA1 signed with 4K RSA you consider insecure these days.  Could you please reconsider your decision? You still have option to give ISV to support customers remaining on unsupported Windows 7/8. Please consider

1. Issue new cross certificate (SHA2+bigger private key) + required dedicated intermediate with private key to third party CA (even with increased requirements on end certificate 4K+ key/shorten validity period of end certificate)

2. new fast guaranteed (within minutes) Windows 7/8 compatible signing by Microsoft

3. cross-sign company dedicated end/intermediate certificate for this purpose 

We started huge complain to convince end customer(s) to upgrade to supported Windows versions few month ago.  Unfortunately, our telemetry analysis shown us zero or insignificant trends in this issue.
I understand this give Microsoft's Trusted Root Program under pressure.

Could you please reconsider mine proposals and give an official statements (Or should we ask for it by different channels on developer.microsoft.com)?

Thank You

Pavol Mišík
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0e326d3e-833a-e312-d35d-11569690c7e6
* Version Independent ID: 7a9795cf-093e-a67f-8162-b832e1161ba8
* Content: [Deprecation of Software Publisher Certificates, Commercial Release Certificates, and Commercial Test Certificates - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates)
* Content Source: [windows-driver-docs-pr/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

pmisik, 2020-11-09:
A few references that documents it affects user mode PE files:

https://social.technet.microsoft.com/wiki/contents/articles/255.forced-integrity-signing-of-portable-executable-pe-files.aspx
> The Forced Integrity policy is enforced on both 32 bit and 64 bit systems and in both **user** and kernel mode. However, Force Integrity is not enforced for boot start drivers.
>    The following Windows features enforce integrity checking on third-party binaries:
>        - Windows Vista and Windows 7: Windows Security Center, Appinit.dll extensions, Object Manager filter registration, and extended process filter registration APIs
>      -  Windows 7: Windows Biometric Framework, LSA extensions, and NTOS kernel extensions

http://download.microsoft.com/download/7/E/7/7E7662CF-CBEA-470B-A97E-CE7CE0D98DC2/AppInit_Win7.docx 
see section Code-sign the DLL with a Authenticode® signature.
>The DLL must be signed with the /ph (page-hash) option. Page-hash enforcement verifies the signature on each page as it is loaded into memory. Because of this requirement, the DLL must be signed on a system that is running Windows Vista or a later version of Windows. The DLL should also be signed with the /t (timestamp) option to prevent the digital signature from expiring when the code-signing certificate expires. **_When you release sign the DLL, you must use the /ac option to reference a cross-certificate. This is required because the signing code path exists in the Windows kernel, which has a hard-coded root of trust. The cross-certificate is required to chain to the issuing certificate authority._**

https://docs.microsoft.com/en-us/archive/blogs/maartenb/performance-impact-of-integritycheck
>It appears that integrity checking bypasses the trusted store. Most likely for performance reasons but that is my guess. **_You need to cross link with a Microsoft certificate._**

tedhudek, 2020-11-11:
Hi @pmisik,

Thanks for the feedback!  We are aware of the problem and are working on it.  I'll keep this issue open and update you when we have a recommended course of action.

tedhudek, 2021-01-22:
Hi @pmisik,

Apologies for the delay.

The recommended method for signing with /IntegrityCheck in the future will be to utilize Azure Code Signing (ACS). However, this service is not yet available, so as an interim solution, MVI partners should request a code signing certificate from Sectigo. 

For more details, please see https://www.yammer.com/wsscengineering/#/files/852228988928.  (You must be an MVI Partner to access that link.)

closed 2021-01-22 by tedhudek