MicrosoftDocs/windows-driver-docs #2559: Bug Check 0xA0 missing details for parameter 1 = 0xf0

GiovanH, 2020-10-26:
Bug Check 0xA0 missing details for parameter 1 = 0xf0

I'm troubleshooting persistent power issues with 

BugcheckCode | 160
BugcheckParameterCode | 0xf0

but the page doesn't list that as a valid parameter code, and doesn't give any instructions about codes not found in the table. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: eb666216-10cc-bf5c-5856-412c08d035ff
* Version Independent ID: 1e42770f-2b01-bf39-e026-5ac39dfd3408
* Content: [Bug Check 0xA0 INTERNAL_POWER_ERROR - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xa0--internal-power-error)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0xa0--internal-power-error.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0xa0--internal-power-error.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-12-05:
@GiovanH - Thanks for this input. I will look at updating 0xA0 with additional parameters. In the mean time you can use `!analyze show 0xA0 0xF0` to see the information provided there.

DOMARS, 2020-12-10:
@GiovanH - Thanks again for this input. I added 0xF0 and a number other newer codes to this topic. Good luck in your work to track down your power issue.

closed 2020-12-10 by DOMARS