MicrosoftDocs/windows-driver-docs #2561: Web page typographic error

HeMinzhang, 2020-10-27:
[image]

https://docs.microsoft.com/en-us/windows/win32/api/fwpstypes/ns-fwpstypes-fwps_filter1

HeMinzhang, 2020-10-27:
[image]

anjoy8, 2020-10-27:
Yes, I found that too.
Browser configuration: Chrome Browser, User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64;  X64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/86.0.4240.111 Safari/537.36

foxdave, 2020-10-27:
up

EliotSeattle, 2020-10-27:
@GrantMeStrength, this one's logged in drivers but the content is in win32. Formatting issue on a ref topic.

Feel free to open a new issue in the win32 repo and point to it from here, and I'll close this.

hsebs, 2020-11-18:
@HeMinzhang, It caused by unpaired tag `</div>`. 
I've fixed the issue, but it takes few days for it to be applied to the page.

GrantMeStrength, 2020-11-18:
Yup, we are experiencing a small delay with the publishing pipeline. Should go live soon. Thanks!

hsebs, 2020-11-21:
@EliotSeattle, It is fixed and applied on page. Could you close this issue?

closed 2020-11-22 by EliotSeattle