MicrosoftDocs/windows-driver-docs #2582: "Hdpi.h Macros" page should be moved

DJm00n, 2020-11-17:
"Hdpi.h Macros" page should be moved from 
`windows-driver-docs` to `windows-driver-docs-ddi` here: https://docs.microsoft.com/windows-hardware/drivers/ddi/hidpi/

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de6e18b3-d937-c3e0-6c8f-947fad282ef8
* Version Independent ID: e3421fcc-dd37-b75c-e1be-41ec9368138b
* Content: [Hdpi.h Macros - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/hdpi-h-macros)
* Content Source: [windows-driver-docs-pr/hid/hdpi-h-macros.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/hid/hdpi-h-macros.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

DJm00n, 2020-11-17:
You can use https://github.com/DJm00n/windows-driver-docs/tree/move-hdpi-h-macros as a base.

billlattimer, 2021-01-25:
@DJm00n, thank you for pointing this out. This article is an orphan in the conceptual documentation, which appears to be an artifact of the list containing mnemonics for other macros defined in the ddi documentation. I'll remove the article and integrate the list into the main index for the HID ddi.

closed 2021-01-25 by billlattimer