MicrosoftDocs/windows-driver-docs #2584: How can I modify the error source address

ginginlay, 2020-11-20:
[Enter feedback here]
How can I modify the error source address in source ?
I did not find where the error source address have been set. 
I would like to try other error addresses in source code and then run exe file.
If there any recommend ways?
Thanks

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 61c73017-14ce-7ec9-208f-95a162a64ba1
* Version Independent ID: bbb12c01-2f4d-1285-ecbc-7b7ab4ad6f73
* Content: [Injecting a Hardware Error - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/injecting-a-hardware-error#feedback)
* Content Source: [windows-driver-docs-pr/whea/injecting-a-hardware-error.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/whea/injecting-a-hardware-error.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-12-03:
Hi @ginginlay, thanks for the feedback.  Your feedback references https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/injecting-a-hardware-error, but I do not see a reference to source address on that page.  Can you clarify your question?

You may have success posting your question in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

tedhudek, 2020-12-09:
Closing out due to inactivity, please feel free to reactivate if still needed

closed 2020-12-09 by tedhudek