MicrosoftDocs/windows-driver-docs #2586: `this` pointer in C++ calling convention is ambiguous

TBBle, 2020-11-21:
Details are in a [Stack Overflow post](https://stackoverflow.com/a/62880979/166389).

The simple problem is
> The calling convention for C++ is very similar: the `this` pointer is passed as an implicit first parameter. The next three parameters are passed in registers, while the rest are passed on the stack.

is ambiguous when compared with the [X64 calling convention for caller-allocated return values](https://docs.microsoft.com/en-us/cpp/build/x64-calling-convention?view=msvc-160&viewFallbackFrom=vs-2019#return-values), where a pointer to the return value storage is _also_ passed as "the first argument".

The ambiguity is the order these rules are applied in.

Per the Stack Overflow post, it seems the resolution observed in MSVC is that the `this` pointer is _always_ RCX, and that hence a return-structure pointer will end up in RDX if both are present.

This ambiguity could be resolved by, for example:
> The calling convention for C++ is very similar: the `this` pointer is passed in RCX. The next three parameters are passed in the remaining registers, while the rest are passed on the stack.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: dc07ad07-7d1c-e42d-506d-de67e4a02494
* Version Independent ID: ed140094-1331-1502-6618-158307cd73f3
* Content: [x64 Architecture - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/x64-architecture#calling-conventions)
* Content Source: [windows-driver-docs-pr/debugger/x64-architecture.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/x64-architecture.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-01-29:
@TBBle - Thanks for this interesting docs issue. I think the wording you proposed is reasonable, and I updated the topic to incorporate that wording.  I also added links to the classic MSDN article, "x64 Primer - Everything You Need To Know To Start Programming 64-Bit Windows Systems" by Matt Pietrek as well as "The history of calling conventions, part 5: amd64" by Raymond Chen. And the "Intel - Introduction to x64 Assembly', which is a nice introduction. As far as what the compiler, and the processor does in it's use of the registers, sequentially in time - compiling code and single stepping through it using WinDbg while viewing the disassembler, and using the register debugger commands, may help clarify that. Thanks again for providing this input and good luck with your debugging activities.

closed 2021-01-29 by DOMARS

TBBle, 2021-02-17:
/reopen

@DOMARS: Looking at the current version of the page (as of 2221b51fe445f7c0b143640bed8bae78bf61ff37), the ambiguity still exists, as the text says

> the `this` pointer is passed as an implicit first parameter

which is ambiguous with the following from [x64 calling convention](https://docs.microsoft.com/en-us/cpp/build/x64-calling-convention?view=msvc-160&viewFallbackFrom=vs-2019#return-values)

> Otherwise, the caller must allocate memory for the return value and pass a pointer to it as the first argument.

Neither of the newly-linked articles clarify that ambiguity either, as [Matt Pietrek](https://docs.microsoft.com/en-us/archive/msdn-magazine/2006/may/x64-starting-out-in-64-bit-windows-systems-with-visual-c) only talked abut the `this` pointer, and [Raymond Chen](https://devblogs.microsoft.com/oldnewthing/20040114-00/?p=41053) only mentions the secret return value pointer, neither addresses when they _both_ want to be the first argument.

As I noted in my original post, by observation of MSVC, in the case that both want to be "the first argument", the rule is that

> the this pointer is passed in RCX.

and then the 'secret first parameter' for a return-value-pointer falls under the next clause.

> The next three parameters are passed in the remaining registers

I think when you used my wording, you missed the first change (about RCX) which was the intended point of the change, and took only the second change (adding 'the remaining', although 'the' was lost, making the sentence a little awkward) which was me just making the sentence read more nicely by qualifying 'registers'.