MicrosoftDocs/windows-driver-docs #2587: Levels not implemented for this platform

gilp456, 2020-11-23:
[Enter feedback here]
Please add some details on the message "Levels not implemented for this platform" when using the !pte.  Ideally, add direction on how to resolve such message.  

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c82454a6-31d6-0cd6-cdd0-f0cea6cea692
* Version Independent ID: 4ffb471c-91cc-5d06-93af-b8ec7c79dc3d
* Content: [pte - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/-pte)
* Content Source: [windows-driver-docs-pr/debugger/-pte.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/-pte.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-01-26:
@gilp456 - I'm not aware of a work around to the  "Levels not implemented for this platform" message when working with !pte. You may be able to use a combination of !syspte, !searchpte, !pfn, along with !vm, !memlist to locate some memory information about the  pte in some situations. You may want to ask about that on the OSR debugger forum with the specifics of what you are trying to locate.  [OSR's windbg List](https://community.osr.com/categories/windbg)

If you believe that the !pte command should be displaying information on your hardware, consider filing product feedback with the debugger team using the https://aka.ms/dexex link.

Thanks for your interest in the debugger docs, and good luck with tracking down the PTE issue you are working on.

closed 2021-01-26 by DOMARS

Ava93, 2021-04-01:
hi gilp456, 
not sure you have solved your problem. I encountered this problem too, and I found this issue may because of low extension version. Finally **update my Windbg** and issue gone. Hope this helps.

xtremegamer1, 2022-09-08:
> 

You are totally unhelpful in resolving this issue so why mark as completed?

mk-ambivert, 2023-08-17:
In my case, it worked to reboot the guest machine several times before getting !pte results (obvious bug in WinDbg).