MicrosoftDocs/windows-driver-docs #2591: Specify download link for WDK

jay-ramani, 2020-11-25:
[Enter feedback here]
While the article starts with WDK, there is no download link for WDK. Would be great if you could make it a bit more convenient.

Thanks,
Jay

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e508dbbe-627c-3ceb-47dd-bfca81044001
* Version Independent ID: c826935e-0d99-1fb2-44b1-9dc598ef1861
* Content: [WDK tasks for MSBuild - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/wdk-tasks-for-msbuild)
* Content Source: [windows-driver-docs-pr/devtest/wdk-tasks-for-msbuild.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/wdk-tasks-for-msbuild.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-11-25:
Thanks for the recommendation, @jay-ramani.

I've added the link and merged to our main branch. It should go live later today.

Cheers,
Eliot

closed 2020-11-25 by EliotSeattle

jay-ramani, 2020-11-25:
Thanks so much for quick response Eliot.

Best,
Jay