MicrosoftDocs/windows-driver-docs #2592: "Dynamic Hardware Partitioning Techniques" has wrong heading in TOC

skissane, 2020-11-27:
In the TOC pane, "Dynamic Hardware Partitioning Techniques" has "Kernel Transaction Manager" as its heading. But I don't think this topic has anything to do with Kernel Transaction Manager. I think this page is meant to start a whole new unrelated section.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5879b1cf-c14b-4be9-59ed-4434e22725b1
* Version Independent ID: 1d7dec5e-5519-2f67-43f6-e4be70939b36
* Content: [Dynamic Hardware Partitioning Techniques - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/dynamic-hardware-partitioning-techniques)
* Content Source: [windows-driver-docs-pr/kernel/dynamic-hardware-partitioning-techniques.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/dynamic-hardware-partitioning-techniques.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

skissane, 2020-11-27:
Here is a screenshot to demonstrate what I mean:
[image]

See how the kernel transaction manager topics come immediately before this topic at the same level. This topic should be one level up, not at the same level as those KTM topics.

tedhudek, 2020-12-02:
Thanks Simon, that's a great find.  Appreciate the screenshot too.

I just pushed the fix.  It should appear on the live site within 10 minutes.

Thanks again!

closed 2020-12-02 by tedhudek