MicrosoftDocs/windows-driver-docs #2593: Page shows option "pnputil /scan-devices", but option is not available

kcmd, 2020-11-28:
The page shows that the following command is valid:

`pnputil /scan-devices`

However, this does not appear to be an available option.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 41c58204-8ae9-7822-df54-6e33af08f3c7
* Version Independent ID: 3ded15a1-1b56-e571-adfd-383b27f64217
* Content: [PnPUtil Examples - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/pnputil-examples?source=docs)
* Content Source: [windows-driver-docs-pr/devtest/pnputil-examples.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/pnputil-examples.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-11-30:
Hi @kcmd  

Can you provide some more info here? I see /scan-devices listed in the tool and I can run the command with that option on my computer.

```
C:\WINDOWS\system32> pnputil /scan-devices
Microsoft PnP Utility

Scanning for device hardware changes.
```

kcmd, 2020-11-30:
The pnputil.exe version shows as 10.0.18362.1 running on Windows 10, version 1903, OS build 18362.1198.

Here's a screen video capture of attempting to issue the command:

https://ufile.io/lg3zr3ym

If that file share method doesn't work then let me know and I'll send it to you another way.

EliotSeattle, 2020-11-30:
Aha! 
The documentation mentions that the /scan-devices option was made available starting in Windows 10 Version 2004, which is Windows build number 10.0.19041. Your build does not yet have /scan-devices incorporated. If you can update to Windows 10 May 2020 Update (also known as version 2004), the functionality will be included.
Hope this helps.
Eliot

closed 2020-11-30 by EliotSeattle

kcmd, 2020-11-30:
That explains it. Thank you!