MicrosoftDocs/windows-driver-docs #2594: Possible Incorrect Description for WFP Packet Inspection Points

vidushv, 2020-11-30:
Under the outgoing packets section, `ALE Connect` and `Transport and ICMP Error` have practically the same description. 

Is the `Transport and ICMP Error` a typo? I am assuming there will be fully reassembled packets under this layer. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: a3e2f6d9-8eff-2de3-38c4-ea49fab06367
* Version Independent ID: 442b5ee2-c76e-eeed-396d-fcabb337e98a
* Content: [Packet Inspection Points - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/packet-inspection-points)
* Content Source: [windows-driver-docs-pr/network/packet-inspection-points.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/packet-inspection-points.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2020-11-30:
Thanks @vidushv for opening this issue. I'll look into this and get back to you.

aviviano, 2021-04-05:
Hi @vidushv. Thanks again for opening this issue. We've updated the description for the Transport and ICMP Error layer (see https://docs.microsoft.com/en-us/windows-hardware/drivers/network/packet-inspection-points).

closed 2021-04-05 by aviviano