MicrosoftDocs/windows-driver-docs #2595: Deprecation of Software Publisher Certificates page is really really vague ?

LusKrew, 2020-12-01:
This page is really, really vague, it doesn't explain in detail how already signed drivers will work or not? there should be a lot more explaining in this page considering how big of a deal this update is..

because there are a lot of applications that are maintained anymore, and they have drivers that have expired certificates but windows right now still loads them so it doesn't break many old applications

now i am asking a simple question : will these drivers with expired certificates that are not maintained anymore by their developers still work or not? because if not then there should be ABSOLUTELY an option for people to not opt in this update and not get this update that can cause many of their old applications to not work anymore. 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0e326d3e-833a-e312-d35d-11569690c7e6
* Version Independent ID: 7a9795cf-093e-a67f-8162-b832e1161ba8
* Content: [Deprecation of Software Publisher Certificates, Commercial Release Certificates, and Commercial Test Certificates - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates)
* Content Source: [windows-driver-docs-pr/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/deprecation-of-software-publisher-certificates-and-commercial-release-certificates.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

YungTanx2, 2020-12-05:
We have the FAQ content below already, what can we add to make it more clear?

**What will happen to my existing signed driver packages?**
As long as driver packages are timestamped before the expiration date of the intermediate certificate, they will continue working.

As for the "update", there is no update. It's that the certificates that Certificate authorities use to issue Software Publisher certs are expiring per the table in the page

nasantia, 2020-12-05:
We have the content below to address the question regarding old unmaintained drivers. Is there anything we can add to make it more clear?

**What will happen to my existing signed driver packages?**
As long as driver packages are timestamped before the expiration date of the intermediate certificate, they will continue working.

As for the "update", there is no update being pushed out. What's happening is that the certificate that Certificate Authorities use to issue Software Publisher certs is expiring.

LusKrew, 2020-12-07:
> We have the content below to address the question regarding old unmaintained drivers. Is there anything we can add to make it more clear?
>
> **What will happen to my existing signed driver packages?** [...]

Ok, so lets say i have a very old application that has also has a driver which has a expired certificate and the driver is not maintained anymore. will this "change" that Microsoft is pushing affect these old type of application and cause their driver to not loader anymore? it would be a lot more helpful if Microsoft was more clear for people who are not really familiar with driver signing and timestamps 

I am asking this is an administrator, because we use a lot of old applications in our network that have drivers with expired certificates (we are the developers of these apps but we use them), i don't want to wake up one day and see Microsoft pushed an updated that causes all of them to stop working.. please be more clear in the MSDN. should i disable Microsoft updates or there is no need to panic?

nasantia, 2020-12-07:
The old applications should continue to work, especially if their certificates are already expired. 

There is no update being pushed out. So even if you were to disable Microsoft update, it wouldn't make a difference. That being said, old applications who rely on old drivers shouldn't be affected by this change.

tedhudek, 2020-12-09:
Your friendly documentation writer here.  I don't see a documentation change to make for this one, so I'm going to close it out.  Please feel free to reactivate as needed

closed 2020-12-09 by tedhudek