MicrosoftDocs/windows-driver-docs #2601: Add details for all arguments (ARG1 ARG2 ARG3 ARG4) for Memory Management 0x1a BSOD

agilp88, 2020-12-04:
[Enter feedback here]
Please add details for what is expected for all 4 arguments for the memory management 0x1a bugcheck, and possible next steps for use in Windbg Preview.  

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: de955911-dfbf-33b3-778a-767a5a578a0d
* Version Independent ID: 23fb5cbd-5f7d-f7ca-1622-5885020966c2
* Content: [Bug Check 0x1A MEMORY_MANAGEMENT - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x1a--memory-management)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x1a--memory-management.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-12-04:
@agilp88 - Not sure I follow. The first parameter indicates the type of memory issue. Depending on the type of memory issue params 2-4 may or may not contain additional data. If they do, that their contents is described after the parameter one value.

DOMARS, 2020-12-18:
@agilp88 - As I don't think there is a docs improvement I can make here, will close this issue. When information is available on param 2-4 it is listed with the associated param 1. For example for 0x1241: *The virtual address for the MDL was unexpectedly asynchronously unmapped midway through the call to build the MDL. Parameter 2 is the MDL, Parameter 3 is the PTE pointer.*
For general information  about troubleshooting bug checks, see [Blue Screen Data](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/blue-screen-data)

closed 2020-12-18 by DOMARS