MicrosoftDocs/windows-driver-docs #2603: Invalid usage example of MmMapIoSpace

I-gor-C, 2020-12-05:
The page provides the following example:

```
MmMapIoSpace(  PhysicalAddress,
        numberOfBytes,
        PAGE_NOCACHE);
```

`MmMapIoSpace` receives `MEMORY_CACHING_TYPE` as the third parameter, which is [defined](https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/wdm/ne-wdm-_memory_caching_type) as:
```
typedef enum _MEMORY_CACHING_TYPE {
  MmNonCached,
  MmCached,
  MmWriteCombined,
  MmHardwareCoherentCached,
  MmNonCachedUnordered,
  MmUSWCCached,
  MmMaximumCacheType,
  MmNotMapped
} MEMORY_CACHING_TYPE;
```
`PAGE_NOCACHE` is not part of this `enum` and is defined as `0x200` in `wdm.h`. It's intended usage is with the `MmMapIoSpaceEx` function, not with `MmMapIoSpace`. In the given example, even if it compiles, `PAGE_NOCACHE` will result in garbage input to the function. Correct example should be something like this:
```
MmMapIoSpace(  PhysicalAddress,
        numberOfBytes,
        MmNonCached);
```

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: cdbcb5a3-69e2-aa09-7311-cdc7dce8c04a
* Version Independent ID: cb040cf8-4b38-7cfe-7918-c08ca0cddc16
* Content: [C30029 - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/30029-memory-allocating-function-requests-executable-memory)
* Content Source: [windows-driver-docs-pr/devtest/30029-memory-allocating-function-requests-executable-memory.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/30029-memory-allocating-function-requests-executable-memory.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

rodwiddowson, 2020-12-05:
Wow! Impressive find!

DOMARS, 2020-12-10:
@I-gor-C - Yes this seems to be a typo, exactly as described. I will work on a PR to fix this. Thanks for the sharp eye here.