MicrosoftDocs/windows-driver-docs #2605: API ERROE

binbibi, 2020-12-07:
[Enter feedback here]
Hello,We Use Hardware dashboard API For Our Dev CI, But https://manage.devcenter.microsoft.com/v2.0/my/hardware/products
got  HTTP 403 recently,Can Fix it?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5954ef94-f961-3260-6a31-b748a5e05bbe
* Version Independent ID: 9755adbb-289b-418d-ef8e-67b6d42991cd
* Content: [Hardware dashboard API - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/dashboard-api#feedback)
* Content Source: [windows-driver-docs-pr/dashboard/dashboard-api.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/dashboard-api.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-12-07:
Hi @binbibi.

Thanks for the feedback. For a 403 error, it likely means you didn’t pass the bearer token as a part of the request. If this does not solve the problem, please open a support ticket. To contact support and open a Hardware Dev Center support case for investigation, **first**, sign into your hardware dev center account, and **then** go to [https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq).

Find: 
 
**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). 

Once there, select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

Hope this helps.
Eliot

closed 2020-12-07 by EliotSeattle

binbibi, 2020-12-08:
@EliotSeattle  thanks