MicrosoftDocs/windows-driver-docs #2620: Driver guidance for WOW64 uses mathematical precision to describe pointers

ianhan, 2020-12-16:
The author of this article seems to have been confused by fixed point/floating point and borrowed "fixed precision" as a term for pointers:
"Windows on Windows (WOW64) enables Microsoft Win32 user-mode applications to run on 64-bit Windows. It does this by intercepting Win32 function calls and converting parameters from pointer-precision types to fixed-precision types as appropriate before making the transition to the 64-bit kernel."

This should be:
"Windows on Windows (WOW64) enables Microsoft Win32 user-mode applications to run on 64-bit Windows. It does this by intercepting Win32 function calls and converting parameters from 32-bit pointer types to 64-bit pointer types as appropriate before making the transition to the 64-bit kernel."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 89391d2e-e972-ae84-e8f6-f85ac21e0dcd
* Version Independent ID: 980565ff-3bd5-98a1-7b46-d1bd1e6e64d7
* Content: [Supporting 32-Bit I/O in Your 64-Bit Driver - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/supporting-32-bit-i-o-in-your-64-bit-driver)
* Content Source: [windows-driver-docs-pr/kernel/supporting-32-bit-i-o-in-your-64-bit-driver.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/supporting-32-bit-i-o-in-your-64-bit-driver.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2020-12-26:
@ianhan . thanks for pointing out. 
I am going to rectify this error.

tedhudek, 2021-01-12:
Thanks, the fix for this will be on the live site momentarily

closed 2021-01-12 by tedhudek