MicrosoftDocs/windows-driver-docs #2628: C# sample has incorrect data

alfaunits, 2020-12-20:
The C# sample indicates that we should use tenantname.onmicrosoft.com instead of {tenant_id} as part of the TokenEndpoint URL (Program.cs, line 27).

Not an easy thing to figure by any means.

---
#### Document details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5954ef94-f961-3260-6a31-b748a5e05bbe
* Version Independent ID: 9755adbb-289b-418d-ef8e-67b6d42991cd
* Content: [Hardware dashboard API - Windows drivers](https://docs.microsoft.com/en-gb/windows-hardware/drivers/dashboard/dashboard-api)
* Content Source: [windows-driver-docs-pr/dashboard/dashboard-api.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/dashboard-api.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-12-21:
Thanks for bringing this to my attention, @alfaunits.

I checked with the team, and they assure me that in this case, either will work. They have scripts that use `{tenantname}.onmicrosoft.com` and others that use `{tenant_id}.` 

However, after looking at the sample, they believe they need to update that to take some new approaches into account. I couldn't get an expected date for that, but hopefully, they'll get to it in short order.

Again, thanks for the Issue.
Eliot

closed 2020-12-21 by EliotSeattle

alfaunits, 2020-12-21:
I tried both, and the tenantname style did not work.
I saw people online had the same issue.

It could be a temporary issue with the website.