MicrosoftDocs/windows-driver-docs #2629: This does uninstall the driver completely, after reboot the appears again in device manager

vignura, 2020-12-24:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 963a0877-81d6-57b1-2922-459c9077c76c
* Version Independent ID: 49998df1-e7fd-6ce7-eb49-aa9c37f13cd8
* Content: [Using Device Manager to Uninstall Devices and Driver Packages - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-device-manager-to-uninstall-devices-and-driver-packages)
* Content Source: [windows-driver-docs-pr/install/using-device-manager-to-uninstall-devices-and-driver-packages.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/using-device-manager-to-uninstall-devices-and-driver-packages.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2020-12-26:
@vignura . it Does not appear after reboot. Because I have tested this method many times in different configurations computers 
if you not select sottware of this device, then driver will be installed again after reboot too.

tedhudek, 2021-01-12:
Thanks for the feedback, @vignura.  Did you select the **Delete the driver software for this device** option?  If so, and this is still an issue, you might check in on the forums to see if anyone there can offer a recommendation.

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

closed 2021-01-12 by tedhudek