MicrosoftDocs/windows-driver-docs #2631: This page should be used for the blue screens as its more useful for troubleshooting.

sebastiannielsen, 2020-12-26:
I find the page https://windows.com/stopcode a totally useless page, that isn't error-dependent, and contains a totally useless guide on how to "fix" the problem.

This page, with the reference, I personally think, its MUCH more useful in actually solving the problem.

My suggestion: Make this page (with customized URLs for the bluescreens) as target for the QR code on the Windows Bugcheck Screens instead. See attached picture on how it would look.

[image]

A easier solution, to make backwards compatible, would be to have an input box on https://windows.com/stopcode where you enter like "CRITICAL_PROCESS_DIED" and then you would get redirected instantly to https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xef--critical-process-died 

also BTW, that blue "Important" box  and the purple "Note" box about the topic being for programmers, should IMHO be removed totally.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ddc632b2-f4e1-b784-dd1e-878feac077f9
* Version Independent ID: 7a6866f4-3333-f2d1-86d7-dbb3d00925e0
* Content: [Bug Check Code Reference - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-code-reference2)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-code-reference2.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-code-reference2.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2021-01-11:
@sebastiannielsen , we cannot give an exact solution for these issues, even windows os designers too.  The reason is if corrupted drivers or incompatible executable process are installed and is not compatible, and is not signed to run in windows 10, then we definitely get critical process error gives.

sebastiannielsen, 2021-01-11:
@RAJU2529 You are completely misunderstanding my issue.

The issue is not that particular bluescreen or any bluescreen.

What im saying, is that on the current bluescreen or Stop Check / Stop Code screen, there is a link to a totally worthless page at http://windows.com/stopcode which isn't helpful at all. I do know how to solve bluescreens, and doesn't have any problems with any bluescreens. This is just to make it more helpful for other computer users.

What im proposing is one of these solutions:
Either that the file responsible for creating the blue screens or stop codes, uses this page (the documentation that this issue is linked to) as link instead of http://windows.com/stopcode

**AND/OR** (preferable AND for backwards compatibility support)

That the http://windows.com/stopcode page, is updated so instead of the worthless guide, it should contain a text input box, where you write like "CRITICAL_PROCESS_DIED" or "INACCESSIBLE_BOOT_DEVICE" or another error code, and then the page should redirect you to for example https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0xef--critical-process-died if you entered CRITICAL_PROCESS_DIED or it could redirect to https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x7b--inaccessible-boot-device if you enter INACCESSIBLE_BOOT_DEVICE in the text box.

This documentation, that the issue refers to, is MUCH MORE HELPFUL than the totally useless http://windows.com/stopcode guide that is more useless than ever.

DOMARS, 2021-01-20:
@sebastiannielsen - Thanks for this input on what guidance we should provide for the stop codes. It is a difficult balance between the different audiences that consume this information. The main customer landing page is pointed at everyone, no matter their level of understanding of Windows and computing. There it covers the few things that all users can do, such as rolling back or checking for driver updates. 

At the next level are IT technical staff, that can look in event logs, understand how drivers are installed, use devcon, etc. There is some guidance for Window Server for example for this group. https://docs.microsoft.com/en-us/windows/client-management/troubleshoot-stop-errors.  

Lastly these pages are directed towards programmers, who own the code that is triggering the stop code, and are working to fix that code, typically with a kernel debugger attached. 

We do are best to balance the guidance for the different types of users, and use input, like git issues, to help direct our limited resources.

On the idea of a customized QR code that would direct users to different stop code specific guidance, you might consider starting a Windows product feedback item, using the feedback app - https://support.microsoft.com/en-us/windows/send-feedback-to-microsoft-with-the-feedback-hub-app-f59187f8-8739-22d6-ba93-f66612949332

Thanks for your interest in the stop code documentation, and good luck with your stop code troubleshooting work.

closed 2021-01-20 by DOMARS

sebastiannielsen, 2021-01-20:
@DOMARS In my opinion, I would say these pages, are more suitable for all audiences, including general consumers, IT technical staff and also programmers, as they contain the most in-depth and best resources to troubleshoot such errors.

These troubleshooting resources also contain very simple and clear descriptions (under the "Cause" header), that everyone can understand and consume.
Thats why I think this documentation should be the main landing page for the https://windows.com/stopcode URL.