MicrosoftDocs/windows-driver-docs #2632: Target needs DelayAPO.dll and KWSApo.dll

aprykhus, 2021-01-02:
I had to copy DelayAPO.dll and KWSApo.dll to target to install driver. Instructions; section 3, step 6; didn't list these files.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9226c2e0-2c63-788b-5164-4856f66cc12b
* Version Independent ID: ba393947-45d1-935d-aba2-b776e163915b
* Content: [Debug Drivers - Step-by-Step Lab (Sysvad Kernel Mode) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/debug-universal-drivers--kernel-mode-)
* Content Source: [windows-driver-docs-pr/debugger/debug-universal-drivers--kernel-mode-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/debug-universal-drivers--kernel-mode-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-01-19:
@aprykhus - Thanks, yes it looks like the built dlls have changed, will update the docs.

DOMARS, 2021-01-21:
@aprykhus - Thanks for seeing this and filing the git docs issue! The topic is now updated to mention the current list of dlls, including DelayAPO.dll and KWSApo.dll.

closed 2021-01-21 by DOMARS