MicrosoftDocs/windows-driver-docs #2640: No information on header/lib requirements.

coolhandle01, 2021-01-13:
There's no information on includes here. Hitting F1 with `EvtNext` told me about  `#include <winevt.h>`, yet yields nothing for `ProcessHwErrorEvent(EventHandle);`.

Where is `ProcessHwErrorEvent(EventHandle);`  or some superceding function defined?

Actually trying to diagnose a hardware problem and I have to compile code. Awesome.
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 759a7b35-c5ac-8ead-d483-3417ab3c01ed
* Version Independent ID: 72f27974-7da4-0154-f9c8-e37eacf337ce
* Content: [Registering for Notification of Hardware Error Events - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/whea/registering-for-notification-of-hardware-error-events)
* Content Source: [windows-driver-docs-pr/whea/registering-for-notification-of-hardware-error-events.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/whea/registering-for-notification-of-hardware-error-events.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

coolhandle01, 2021-01-13:
https://docs.microsoft.com/en-us/search/?terms=ProcessHwErrorEvent = No Results.

tedhudek, 2021-01-21:
Thanks for the feedback.

In that example, `ProcessHwErrorEvent` is just an example function that the driver would implement.

To debug a hardware problem, you probably want to look for the WHEA-Logger event in the system event log. This appears in a human readable format and should not require any code. 

Best of luck!

closed 2021-01-21 by tedhudek

coolhandle01, 2021-01-21:
Cheers! AMD Drivers.. unreal the trouble they cause me!

tedhudek, 2021-01-21:
Try this --

[image]