MicrosoftDocs/windows-driver-docs #2648: Parameters' description

LuigiBruno, 2021-01-18:
Provide a description of each of the parameters for this bugcheck.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ec8f80f5-7104-5fe9-8f10-adf04f6c6a8c
* Version Independent ID: c087aec7-b173-a4d0-4b23-742982c736e7
* Content: [Bug Check 0x20001 HYPERVISOR_ERROR - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/bug-check-0x20001--hypervisor-error)
* Content Source: [windows-driver-docs-pr/debugger/bug-check-0x20001--hypervisor-error.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/bug-check-0x20001--hypervisor-error.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2021-01-29:
@LuigiBruno - Thanks for the input on providing more information on the Bug Check 0x20001 HYPERVISOR_ERROR. It looks like this will more frequently be a bug check that is more related to the internal operation on the hypervisor. So at this point, will not be providing more information on the parameters.  As I have mentioned before, we use a process to prioritize the most frequent bug checks out of the hundreds that are currently defined. With our limited documentation team resources, it is likely that we will only be able to do this work for a small sub set of all of the possible stop codes that have broad applicability. Again, thanks for the input, and good luck with your Windows troubleshooting.

closed 2021-01-29 by DOMARS

LuigiBruno, 2021-02-22:
> @LuigiBruno - Thanks for the input on providing more information on the Bug Check 0x20001 HYPERVISOR_ERROR. It looks like this will more frequently be a bug check that is more related to the [...]

OK @DOMARS.
Thank you.