MicrosoftDocs/windows-driver-docs #2649: Sign driver for internal distribution

Stuv7CB, 2021-01-20:
Hello.

Could you please clarify section [Signing a driver for internal distribution only](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later-#signing-a-driver-for-internal-distribution-only)?

* Is it only for Windows 10 or older versions are supported?
* Is it applicable for Windows Server?
* Is it mandatory to sign a driver with non-EV codesign cert?
* Is there a step-by-step guide for submitting such package in hardware dashboard portal? I managed to find uploading regular .hlkx package, wlk package and .cab file for attestation signature only

Thank you in advance

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 1eb19d33-8f4c-f379-b645-938a746c5615
* Version Independent ID: 8efb9fb4-1e9a-1a8a-994b-450495449e15
* Content: [Driver Signing Policy - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/kernel-mode-code-signing-policy--windows-vista-and-later-#signing-a-driver-for-internal-distribution-only)
* Content Source: [windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/kernel-mode-code-signing-policy--windows-vista-and-later-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2021-01-21:
Hi Leonid - that procedure applies to all supported versions of Windows.  I just added a note to the doc page to that effect. It is applicable to Server if you are doing full HCK/HLK signing, as attestation doesn't apply to Server.

For a step-by-step, I would start with https://docs.microsoft.com/windows-hardware/drivers/dashboard/create-a-new-hardware-submission.

I'm waiting to hear back if it's mandatory to sign with a non-EV codesign cert.

tedhudek, 2021-01-21:
It is mandatory to sign with a non-EV codesign cert.  

Thanks for the feedback, and best of luck!

closed 2021-01-21 by tedhudek

Stuv7CB, 2021-01-22:
@tedhudek thank you for your response! You have said that
> It is applicable to Server if you are doing full HCK/HLK signing, as attestation doesn't apply to Server.

The article describes that one can upload .cab with driver package, not .hlkx with test logs for internal signing. Is it obligatory to run HCK/HLK tests for windows server (including windows server 2012, 2012R2)?

And just to clarify
> It is mandatory to sign with a non-EV codesign cert.
If I have a driver with EV sign I can't sign it for internal distribution?

tedhudek, 2021-01-22:
Sorry, I should clarify - you do need to run HCK/HLK tests for Windows Server 2016 and later.  

And you can use either EV or non-EV codesign but the cert must be registered in Partner Center before you upload the package.