MicrosoftDocs/windows-driver-docs #2652: ?  not fount TDR registery entry in windows 10 ...

rogermachin, 2021-01-21:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 652e03eb-a834-6c30-44c1-e9c917a94926
* Version Independent ID: bca54aa8-2da8-e389-aa74-f66bf3ab65ab
* Content: [Testing and debugging TDR - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/tdr-registry-keys)
* Content Source: [windows-driver-docs-pr/display/tdr-registry-keys.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/tdr-registry-keys.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

rogermachin, 2021-01-22:
i  add the two  missing entry in regedit , that's seem ok  :
[HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\GraphicsDrivers]

"TdrDelay"=dword:0000003c
"TdrDdiDelay"=dword:0000003c

RAJU2529, 2021-01-25:
@rogermachin 
this article  author is cleared  included note 
**These registry keys should not be manipulated by end users, or by applications outside of targeted testing or debugging during driver development.**
If you are end-user of windows 10, then you can add the registry keys for debugging process, not the regular process.

RAJU2529, 2021-01-25:
@lorihollasch , I  verified the article  in the edit process, and i found One  table  format is  included in source content , but that table is not reflected in public view in website

RAJU2529, 2021-01-25:
@e0i  @nam31   @lorihollasch, please  verify PR #2655 and approve if it correct.

**https://github.com/MicrosoftDocs/windows-driver-docs/pull/2655**

lorihollasch, 2021-01-25:
@rogermachin , this page (and everything under https://docs.microsoft.com/en-us/windows-hardware/drivers) is intended for hardware/software driver developers, and not for end users. Per this page, we do not recommend that the registry edits be done by end users or apps.

closed 2021-01-25 by lorihollasch

rawwar, 2021-03-09:
@RAJU2529, @lorihollasch  , I understand that it's not intended for manipulation by the end-user. But, if I were to check the value of TDRdelay, how should I do it?

rawwar, 2021-03-09:
Yes, I know that. My main intention was to get the timeout values and show them to the user.

RAJU2529, 2021-03-09:
@rawwar . Sorry that's not possible.

rogermachin, 2021-03-15:
thanks for your's answer  .

bests regards ,

Roger